r-darwish/alma #19: bsdcpio: Failed to set default locale

anibal-aguila, 2019-05-27:
Hi, I get this error... how I can fix it?

```
==> Creating gzip-compressed initcpio image: /boot/initramfs-linux-fallback.img
bsdcpio: Failed to set default locale
==> Image generation successful
( 3/13) Warn about old perl modules
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
        LANGUAGE = (unset),
        LC_ALL = (unset),
        LC_MESSAGES = "",
        LC_CTYPE = "en_US.UTF-8",
        LANG = "en_US.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to the standard locale ("C").

```

thanks in advance,

r-darwish, 2019-05-27:
What's the output of `locale` in your system?

anibal-aguila, 2019-05-27:
❯ locale

```
LANG=en_US.UTF-8
LC_CTYPE=en_US.UTF-8
LC_NUMERIC="en_US.UTF-8"
LC_TIME="en_US.UTF-8"
LC_COLLATE="en_US.UTF-8"
LC_MONETARY="en_US.UTF-8"
LC_MESSAGES=
LC_PAPER="en_US.UTF-8"
LC_NAME="en_US.UTF-8"
LC_ADDRESS="en_US.UTF-8"
LC_TELEPHONE="en_US.UTF-8"
LC_MEASUREMENT="en_US.UTF-8"
LC_IDENTIFICATION="en_US.UTF-8"
LC_ALL=

```

r-darwish, 2019-05-27:
I ran the command and I see the error as well, but the command doesn't fail and `mkinitcpio` runs as expected. I don't think I can fix it because it happens in the `pacstrap`, before I can even set a locale.

anibal-aguila, 2019-05-27:
I handle it by running after the installation this script:

alma-postInstall.sh

```
#!/bin/bash

echo 'en_US.UTF-8 UTF-8' >> /etc/locale.gen
locale-gen

ln -s /dev/null /etc/udev/rules.d/80-net-setup-link.rules
cp /etc/netctl/examples/ethernet-dhcp /etc/netctl/eth0-arch_usb
rm /etc/systemd/system/multi-user.target.wants/NetworkManager.service
```

Even I remove NetworkManager service link because wifi-menu don't want to start if it's active.

commit 0492847d (2019-05-27): Generate locale (fix #19)
closed 2019-05-27 by r-darwish