r-darwish/alma #22: sgdisk and binary path

testbughub, 2019-08-02:
Hi!

I'm not sure if I'm using it correctly, but when I type in:
./alma create /dev/disk/by-id/usb-VendorCo_ProductCode_920710705A405112181-0\:0
I get this error:
 2019-08-02 08:44:45 ERROR alma > Could not find sgdisk
 2019-08-02 08:44:45 ERROR alma > Caused by: Cannot find binary path

I'm a newbie at Arch, so forgive me if I made a simple mistake.

r-darwish, 2019-08-02:
Did you build the application from source? If so, you'll need to install sgdisk manually

testbughub, 2019-08-02:
I may have misunderstood how this application is supposed to be used.
I managed to install sgdisk, but now it says:
 2019-08-02 08:53:03 ERROR alma > Could not find pacstrap
 2019-08-02 08:53:03 ERROR alma > Caused by: Cannot find binary path

r-darwish, 2019-08-02:
You can take a look at Dockerfile to see the dependencies required by the application, but I suggest that you install it from AUR using yay or some other AUR helper, as it will take care installing the required dependencies. Building from source is recommended only for developers

testbughub, 2019-08-02:
True.
I'll dive into some more guides before posting unnecessary issues :)
Thanks for taking time to help this newbie!

closed 2019-08-02 by testbughub