r-darwish/alma #25: GRUB issues and documentation

jamesmcm, 2019-11-24:
Hi,

This tool is amazing, and saved me loads of time! I added it to the ArchWiki page: https://wiki.archlinux.org/index.php?title=Installing_Arch_Linux_on_a_USB_key as I had just found it by searching on GitHub.

I had a few minor issues:

- I had to reinstall GRUB to get MBR booting working, I'm not sure why, but reinstalling it worked perfectly ([per this guide](http://valleycat.org/linux/arch-usb.html)). I had formatted the drive to an unformatted partition before starting, and alma seemed to treat that as though it were the existing MBR partition. I'm not sure if that was the cause of the issue or not.

- In my case, I set up an encrypted home partition with dm-crypt. This is quite hard to get right (I still don't have locking after unmount working correctly on logout), so it'd be great if that could be added (as well as a full disk encryption option perhaps).

- I set up [libeatmydata](https://wiki.archlinux.org/index.php?title=Installing_Arch_Linux_on_a_USB_key&oldid=589905#Minimizing_disk_access) and [profile-sync-daemon](https://wiki.archlinux.org/index.php/Profile-sync-daemon) for Chromium. It'd be cool to have this as an option to install automatically (or in a comprehensive preset example).

- I would still like to set it up to work with Secure Boot. This would be another great option for the installer.

But overall I think it'd be great to add a user guide to the README.md and some more comprehensive presets. As I did some work myself setting up dm-crypt and then noticed you have crypt.rs in the source code handling some LUKS stuff, so maybe alma can do it directly.

The guide could also include installing yay by default in a preset, and exporting the package list from an existing Arch system, how to migrate dotfiles, etc.

Overall this tool is incredible, and I think with a little work on the documentation and included presets, it could become a standard piece of the Arch Linux ecosystem.

r-darwish, 2019-11-24:
> * I had to reinstall GRUB to get MBR booting working, I'm not sure why, but reinstalling it worked perfectly ([per this guide](http://valleycat.org/linux/arch-usb.html)). I had formatted the drive [...]

I'm not sure what's causing it. Does `dd` your device with zero before running ALMA helps?

>     * In my case, I set up an encrypted home partition with dm-crypt. This is quite hard to get right (I still don't have locking after unmount working correctly on logout), so it'd be great if that could be added (as well as a full disk encryption option perhaps).

Full disk encryption is supported with the `-e` switch.

> *
>
> * I set up [libeatmydata](https://wiki.archlinux.org/index.php?title=Installing_Arch_Linux_on_a_USB_key&oldid=589905#Minimizing_disk_access) and [...]

You can write presets file and have ALMA configure your device based on them. See [this example](https://github.com/r-darwish/alma/blob/master/presets/user.toml).

>     * I would still like to set it up to work with Secure Boot. This would be another great option for the installer.

That would be awesome. Although, I don't have the hardware to test it nor the knowledge how to do it.

> Overall this tool is incredible, and I think with a little work on the documentation and included presets, it could become a standard piece of the Arch Linux ecosystem.

Thank you! This tool could really use some more documentation.

jamesmcm, 2020-03-21:
Closing, as the documentation is done in #28 , created issue #44 regarding Secure Boot.

I'll test the MBR booting now I have a new USB drive too (but I want to finish creating some presets first).

Thanks for the great tool!

closed 2020-03-21 by jamesmcm