r-darwish/alma #30: No check to see if grub-install is present

cedws, 2020-03-08:
Hey, just a quick thing I spotted whilst looking through the code. There's already some checks in place to see if various tools are present, but there is no such check for `grub-install`. Most people are going to have GRUB installed anyway, so no big deal.

https://github.com/r-darwish/alma/blob/19eef3a0e1efad867d7a010ae2f22b319a598c61/src/main.rs#L125-L140

jamesmcm, 2020-03-09:
I don't think this is a problem, since `grub-install` is only executed inside the chroot. And grub is in the base packages always installed by ALMA, see:

https://github.com/r-darwish/alma/blob/19eef3a0e1efad867d7a010ae2f22b319a598c61/src/main.rs#L226-L233

And note that it is only used as an "argument" inside the chroot:

https://github.com/r-darwish/alma/blob/19eef3a0e1efad867d7a010ae2f22b319a598c61/src/main.rs#L354-L360

`pacstrap` will use the cached version on the host if it exists, but I think it will download it if it's not installed.

r-darwish, 2020-03-10:
Good explanation by @jamesmcm

closed 2020-03-10 by r-darwish

cedws, 2020-03-11:
Makes sense, thanks.