r-darwish/alma #31: Preset / Initramfs error

jsuhm, 2020-03-13:
I tried to create an encrypted live usb but there was an error:

$ sudo alma -v create -e /dev/sdb 
...
/usr/bin/mkinitcpio: line 265: /etc/mkinitcpio.d/linux.preset: No such file or directory
==> ERROR: Failed to load preset: `/etc/mkinitcpio.d/linux.preset'
 2020-03-13 01:37:49 DEBUG alma::storage::mount_stack    > Unmounting /tmp/.tmpHsd2Xf/boot
 2020-03-13 01:37:51 DEBUG alma::storage::mount_stack    > Unmounting /tmp/.tmpHsd2Xf
 2020-03-13 01:38:14 DEBUG alma::storage::crypt          > Closing encrypted device alma_root
 2020-03-13 01:38:14 ERROR alma                          > Initramfs error
 2020-03-13 01:38:14 ERROR alma                          > Caused by: exit code: 1

r-darwish, 2020-03-13:
Maybe something changed with the installation procedure. I don't have as much time as I used to to maintain this tool, but I'll try to take a look

jamesmcm, 2020-03-21:
Looking here: https://wiki.archlinux.org/index.php/Mkinitcpio#Image_creation_and_activation

>  The presets will be removed from the kernel packages. mkinitcpio 27 uses a pacman hook that creates the presets from a template (/usr/share/mkinitcpio/hook.preset).

It looks like the presets aren't included by default anymore.

I'll try to take a look, but it's working fine on my system.

@jsuhm is your host a new install of Arch Linux?

jamesmcm, 2020-03-21:
Hopefully https://github.com/r-darwish/alma/pull/36 will fix this, please test it.

closed 2020-03-21 by r-darwish