r-darwish/alma #32: not working in Manjaro

yananet, 2020-03-16:
Hello

thank you for the great tool.
I am not sure whether manjaro is supported or not anyway it is not working with my manjaro laptop
```
Generating locales...
  en_US.UTF-8... done
Generation complete.
 2020-03-17 07:45:16 INFO  alma > Generating initramfs
/usr/bin/mkinitcpio: line 265: /etc/mkinitcpio.d/linux.preset: No such file or directory
==> ERROR: Failed to load preset: `/etc/mkinitcpio.d/linux.preset'
 2020-03-17 07:45:22 ERROR alma > Initramfs error
 2020-03-17 07:45:22 ERROR alma > Caused by: exit code: 1
```

r-darwish, 2020-03-17:
I don't officially support Manjaro. However, this looks like #31 so it's probably an Arch Linux issue as well, so I'm closing this as duplicate

closed 2020-03-17 by r-darwish

yananet, 2020-03-18:
FYI
was worked in my Arch desktop.

jamesmcm, 2020-03-21:
Hopefully #36 will fix this, please test it.