r-darwish/alma #40: Add flag to disable LiveUSB optimisations

jamesmcm, 2020-03-21:
## Task

Add a flag to disable LiveUSB optimisations, allowing ALMA to also function as install scripts for a standard system on a hard disk (we could even consider adding choices of SSD optimisations, etc. but that is probably better left to preset files).

## Justification

This would allow the user to create a USB with ALMA to try out their installation from their preset files. Then if they like it they could install it to disk from that very USB distribution.

r-darwish, 2020-03-21:
I don't object, but I should note that focus of this tool will keep being for LiveUSBs. I'm not looking for creating a generic installer.

DwarfFighterCleric, 2020-11-28:
Replying to James (OP), maybe what you're looking for is something like JLiveCD?  
It uses the Chroot method to modify a Live-Session ISO, in the sense that it can allow you for example to deploy multiple machines without having to do any additional tweaking, all the work has been done in the 1st step of creating the ISO.  
The bad news is: it doesn't have Manjaro support yet, just pure Arch and Ubuntu-based distros.  
I'm looking for ways to find a similar Chroot method that works for Manjaro. Because I noticed Manjaro splits the Live-Session into multiple separate SquashFS's, instead of using a single SquashFS like other distros do. I wonder what differences are in the way they are mounted or why that is done.