r-darwish/alma #43: Add aur_helper and aur_packages fields to presets

jamesmcm, 2020-03-21:
## Task

Add `aur_helper` and `aur_packages` fields to presets. 

The aur_helper will need to be installed separately (and this is not trivial, maybe this should be built in to ALMA). 

The aur_helper will also need to be customisable (since the "best" helper has changed often i.e. from yaourt to pacaur to yay).

## Justification

This allows preset developers to install packages via the AUR without having to do it in custom scripts (which are harder for the users to review).

r-darwish, 2020-03-21:
I thought about this and didn't come up with any solution I'm happy with. This would be a good feature but I still debate about how to implement it.

jamesmcm, 2020-03-21:
Yeah, the tricky thing is handling different AUR helpers, but also not making the presets depend on one another.

Maybe the user can pass their AUR helper to ALMA which ALMA installs, and then the presets just specify the AUR packages to install (if any).

closed 2020-06-18 by r-darwish

txtsd, 2022-01-28:
Could you make a new release so we can have access to the --aur-helper flag?