r-darwish/alma #44: Create preset example for Secure Boot

jamesmcm, 2020-03-21:
## Task

Create a preset file that works for booting with Secure Boot enabled.

Or we could override the boot loader directly in this case.

See: https://wiki.archlinux.org/index.php/Secure_Boot#Using_a_signed_boot_loader for details

## Justification

Would allow an ALMA-created USB to boot on even more machines (i.e. when we can't disable Secure Boot).

r-darwish, 2020-03-21:
Actually I rather change the tool itself so that it supports secure boot as well. Sadly I don't have the hardware to test it or the knowledge to do that, so help is welcome.

r-darwish, 2020-11-26:
@jamesmcm do you have the hardware to test it? Can you take an existing ALMA image and manually install PreLoader and test that it's working? If it does, then I think we can always install preload in ALMA so there won't be any need to using a preset just to enable secure boot, but I will need someone to test it out.

jamesmcm, 2020-11-26:
Sure, I'll try it when I get some time in December. I'll need to re-enable the Secure Boot keys and make sure I can boot my main system 🙂

commit 6924221b (2020-11-30): Enable secure boot (fix #44) (#65)
closed 2020-11-30 by r-darwish