r-darwish/alma #47: grub fails to install

kris004, 2020-04-04:
Thank you for this project. Had one issue, when the usb has a existing label grub fails to install. For example, if your usb has a fs type of iso9660 with a fs label from imaging an archlinux install iso onto the drive grub install errors. Fdisk will remove this by default, gdisk will not. Possibly related to #25.

Best,
Kris

jamesmcm, 2020-04-09:
Thanks for sharing this, do you have more details on the error?

I have a spare USB stick I can try to do some tests on.

r-darwish, 2020-11-27:
Closing due to inactivity

closed 2020-11-27 by r-darwish

sabitm, 2021-02-09:
Hi, i'm experiencing the same issue. It looks like the problem arose from the previous partition label. And can be easily fixed by deleting all partition (using fdisk/cfdisk) and then re-executing alma. Hope this helps some experiencing the same issue