r-darwish/alma #49: Installation fails while generating initramfs

Dorifer, 2020-05-18:
tried alma on a newly formatted usb stick, the installation stopped here: 

dorifer@dorifer-x220:~$ sudo alma create -e
Select a removable device: General UDisk (14.65 GiB)
 2020-05-18 09:31:58 INFO  alma > Partitioning the block device

***************************************************************
Found invalid GPT and valid MBR; converting MBR to GPT format
in memory. 
***************************************************************

GPT data structures destroyed! You may now partition the disk using fdisk or
other utilities.
The operation has completed successfully.
 2020-05-18 09:32:01 INFO  alma > Formatting filesystems
mkfs.fat 4.1 (2017-01-24)
 2020-05-18 09:32:05 INFO  alma > Encrypting the root filesystem
Geben Sie die Passphrase für »/dev/sdb3« ein: 
Geben Sie die Passphrase für »/dev/sdb3« ein: 
mke2fs 1.45.6 (20-Mar-2020)
Ein Dateisystem mit 3813371 (4k) Blöcken und 954720 Inodes wird erzeugt.
UUID des Dateisystems: 532858b9-7657-471b-b7cc-de161335b0a4
Superblock-Sicherungskopien gespeichert in den Blöcken: 
	32768, 98304, 163840, 229376, 294912, 819200, 884736, 1605632, 2654208

beim Anfordern von Speicher für die Gruppentabellen: erledigt                        
Inode-Tabellen werden geschrieben: erledigt                        
Das Journal (16384 Blöcke) wird angelegt: erledigt
Die Superblöcke und die Informationen über die Dateisystemnutzung werden
geschrieben: erledigt

 2020-05-18 09:33:33 INFO  alma > Mounting filesystems to /tmp/.tmphJAQfA
 2020-05-18 09:33:36 INFO  alma > Bootstrapping system
==> Creating install root at /tmp/.tmphJAQfA
==> Installing packages to /tmp/.tmphJAQfA
:: Synchronisiere Paketdatenbanken...
 core                  168,7 KiB  1534 KiB/s 00:00 [######################] 100%
 extra                2010,6 KiB  3,00 MiB/s 00:01 [######################] 100%
 community               5,9 MiB  3,49 MiB/s 00:02 [######################] 100%
 multilib              192,9 KiB  4,71 MiB/s 00:00 [######################] 100%
:: Es stehen 11 Provider für broadcom-wl zur Verfügung:
:: Repositorium extra
   1) linux316-broadcom-wl  2) linux414-broadcom-wl  3) linux419-broadcom-wl
   4) linux44-broadcom-wl  5) linux49-broadcom-wl  6) linux54-broadcom-wl
   7) linux55-broadcom-wl  8) linux56-broadcom-wl  9) linux57-broadcom-wl
:: Repositorium community
   10) linux54-rt-broadcom-wl  11) linux56-rt-broadcom-wl

Geben Sie eine Zahl ein (Voreinstellung=1): 
Löse Abhängigkeiten auf...
Suche nach in Konflikt stehenden Paketen...
Warnung: Abhängigkeits-Zyklus entdeckt:
Warnung: bashrc-manjaro wird vor seiner Abhängigkeit bash installiert werden

Pakete (157) acl-2.2.53-3  archlinux-keyring-20200422-1  argon2-20190702-3
             attr-2.4.48-3  audit-2.8.5-6  bash-5.0.016-1
             bashrc-manjaro-5.0.016-1  bluez-libs-5.54-2  bzip2-1.0.8-3
             ca-certificates-20181109-3  ca-certificates-mozilla-3.52-1
             ca-certificates-utils-20181109-3  coreutils-8.32-1
             cracklib-2.9.7-2  cryptsetup-2.3.2-1  curl-7.70.0-1  db-5.3.28-5
             dbus-1.12.16-5  device-mapper-2.02.187-2  diffutils-3.7-3
             e2fsprogs-1.45.6-2  efivar-37-3  expat-2.2.9-3  file-5.38-3
             filesystem-2019.10-4  findutils-4.7.0-2  gawk-5.1.0-1
             gcc-libs-9.3.0-1  gdbm-1.18.1-3  gettext-0.20.2-1  glib2-2.64.2-1
             glibc-2.31-2  gmp-6.2.0-1  gnupg-2.2.20-1  gnutls-3.6.13-2
             gpgme-1.13.1-5  gpm-1.20.7.r27.g1fd1941-2  grep-3.4-1  gzip-1.10-3
             hwids-20200306-1  iana-etc-20200428-1  icu-67.1-1
             iproute2-5.6.0-1  iptables-1:1.8.4-1.1  iputils-20190709-2
             jansson-2.12-2  js60-60.9.0-2  json-c-0.14-4  kbd-2.2.0-5
             keyutils-1.6.1-4  kmod-27-1  krb5-1.18-1  less-551-3
[... 510 lines cut ...]
::Singapore       : https://download.nus.edu.sg/mirror/manjaro/stable
::China           : https://mirrors.shuosc.org/manjaro/stable
::Portugal        : http://manjaro.barata.pt/stable
::Iran            : https://repo.sadjad.ac.ir/manjaro/stable
::Russia          : http://mirror.truenetwork.ru/manjaro/stable
::Indonesia       : http://kambing.ui.ac.id/manjaro/stable
::Bulgaria        : http://manjaro.telecoms.bg/stable
::Germany         : https://mirror.netzspielplatz.de/manjaro/packages/stable
::INFO Mirror list generated and saved to: /etc/pacman.d/mirrorlist
hint: use `pacman-mirrors` to generate and update your pacman mirrorlist.
(15/15) Rebuilding certificate stores...
 2020-05-18 09:53:55 INFO  alma > Performing post installation tasks
Created symlink /etc/systemd/system/multi-user.target.wants/NetworkManager.service → /usr/lib/systemd/system/NetworkManager.service.
Created symlink /etc/systemd/system/dbus-org.freedesktop.nm-dispatcher.service → /usr/lib/systemd/system/NetworkManager-dispatcher.service.
Created symlink /etc/systemd/system/network-online.target.wants/NetworkManager-wait-online.service → /usr/lib/systemd/system/NetworkManager-wait-online.service.
 2020-05-18 09:53:55 INFO  alma > Configuring journald
 2020-05-18 09:53:55 INFO  alma > Setting locale
Generating locales...
  en_US.UTF-8... done
Generation complete.
 2020-05-18 09:54:05 INFO  alma > Generating initramfs
/usr/bin/mkinitcpio: line 265: /etc/mkinitcpio.d/linux.preset: No such file or directory
==> ERROR: Failed to load preset: `/etc/mkinitcpio.d/linux.preset'
 2020-05-18 09:55:23 ERROR alma > Initramfs error
 2020-05-18 09:55:23 ERROR alma > Caused by: exit code: 1

jamesmcm, 2020-05-28:
You need to install the base package on your host: https://www.archlinux.org/news/base-group-replaced-by-mandatory-base-package-manual-intervention-required/

```
sudo pacman -S linux base
```

Dorifer, 2020-05-29:
Installed the base package, installation hangs itself at the same point (I'm on manjaro if that makes a difference)

jamesmcm, 2020-05-29:
Please try running it with the `--verbose` flag and post the output here:

```
sudo alma -v create -e
```

Also note you need both `linux` and `base`. I will add these as dependencies to the AUR package too.

Dorifer, 2020-05-31:
The output with -v:

Select a removable device: General UDisk (14.65 GiB)
 2020-05-31 09:00:54 DEBUG alma::storage::storage_device > path: "/dev/sdc"
 2020-05-31 09:00:54 DEBUG alma::storage::storage_device > real path: "/dev/sdc", device name: "sdc"
 2020-05-31 09:00:54 DEBUG alma::storage::storage_device > Reading: "/sys/block/sdc/removable"
 2020-05-31 09:00:54 DEBUG alma::storage::storage_device > "/sys/block/sdc/removable" -> 1

 2020-05-31 09:00:54 INFO  alma                          > Partitioning the block device
 2020-05-31 09:00:54 DEBUG alma                          > "/dev/sdc"
Warning: The kernel is still using the old partition table.
The new table will be used at the next reboot or after you
run partprobe(8) or kpartx(8)
GPT data structures destroyed! You may now partition the disk using fdisk or
other utilities.
Warning: The kernel is still using the old partition table.
The new table will be used at the next reboot or after you
run partprobe(8) or kpartx(8)
The operation has completed successfully.
 2020-05-31 09:01:03 INFO  alma                          > Formatting filesystems
 2020-05-31 09:01:03 DEBUG alma::storage::storage_device > Partition 1 for sdc is in "/dev/sdc1"
mkfs.fat 4.1 (2017-01-24)
 2020-05-31 09:01:04 DEBUG alma::storage::storage_device > Partition 3 for sdc is in "/dev/sdc3"
 2020-05-31 09:01:04 INFO  alma                          > Encrypting the root filesystem
 2020-05-31 09:01:04 DEBUG alma::storage::crypt          > Preparing encrypted device in /dev/sdc3
Geben Sie die Passphrase für »/dev/sdc3« ein: 
Gerät /dev/sdc3 wird gerade benutzt. Das Formatieren ist gerade nicht möglich.
 2020-05-31 09:01:21 ERROR alma                          > Error setting up an encrypted device
 2020-05-31 09:01:21 ERROR alma                          > Caused by: exit code: 1
dorifer@dorifer-x220:~$ sudo alma -v create -e
Select a removable device: General UDisk (14.65 GiB)
 2020-05-31 09:04:01 DEBUG alma::storage::storage_device > path: "/dev/sdc"
 2020-05-31 09:04:01 DEBUG alma::storage::storage_device > real path: "/dev/sdc", device name: "sdc"
 2020-05-31 09:04:01 DEBUG alma::storage::storage_device > Reading: "/sys/block/sdc/removable"
 2020-05-31 09:04:01 DEBUG alma::storage::storage_device > "/sys/block/sdc/removable" -> 1

 2020-05-31 09:04:01 INFO  alma                          > Partitioning the block device
 2020-05-31 09:04:01 DEBUG alma                          > "/dev/sdc"
GPT data structures destroyed! You may now partition the disk using fdisk or
other utilities.
The operation has completed successfully.
 2020-05-31 09:04:07 INFO  alma                          > Formatting filesystems
 2020-05-31 09:04:07 DEBUG alma::storage::storage_device > Partition 1 for sdc is in "/dev/sdc1"
mkfs.fat 4.1 (2017-01-24)
 2020-05-31 09:04:08 DEBUG alma::storage::storage_device > Partition 3 for sdc is in "/dev/sdc3"
 2020-05-31 09:04:08 INFO  alma                          > Encrypting the root filesystem
 2020-05-31 09:04:08 DEBUG alma::storage::crypt          > Preparing encrypted device in /dev/sdc3
Geben Sie die Passphrase für »/dev/sdc3« ein: 
 2020-05-31 09:04:22 DEBUG alma::storage::crypt          > Opening encrypted device /dev/sdc3 as alma_root
Geben Sie die Passphrase für »/dev/sdc3« ein: 
mke2fs 1.45.6 (20-Mar-2020)
Ein Dateisystem mit 3813371 (4k) Blöcken und 954720 Inodes wird erzeugt.
UUID des Dateisystems: 55c64c4f-50a8-48c7-a099-0f8676e4c4a8
Superblock-Sicherungskopien gespeichert in den Blöcken: 
	32768, 98304, 163840, 229376, 294912, 819200, 884736, 1605632, 2654208

beim Anfordern von Speicher für die Gruppentabellen: erledigt                        
Inode-Tabellen werden geschrieben: erledigt                        
Das Journal (16384 Blöcke) wird angelegt: erledigt
Die Superblöcke und die Informationen über die Dateisystemnutzung werden
geschrieben: erledigt

 2020-05-31 09:05:34 DEBUG alma                          > Root partition: /dev/mapper/alma_root
 2020-05-31 09:05:34 INFO  alma                          > Mounting filesystems to /tmp/.tmptABGD1
 2020-05-31 09:05:34 DEBUG alma::storage::mount_stack    > Mounting Filesystem { fs_type: Ext4, block: EncryptedDevice { cryptsetup: Tool { exec: "/usr/bin/cryptsetup" }, name: "alma_root", path: "/dev/mapper/alma_root", origin: PhantomData } } to "/tmp/.tmptABGD1"
 2020-05-31 09:05:34 DEBUG alma::storage::mount_stack    > Mounting Filesystem { fs_type: Vfat, block: Partition { path: "/dev/sdc1", origin: PhantomData } } to "/tmp/.tmptABGD1/boot"
 2020-05-31 09:05:37 DEBUG alma                          > lsblk:
NAME                                  FSTYPE      FSVER LABEL       UUID                                 FSAVAIL FSUSE% MOUNTPOINT
sda                                                                                                                     
├─sda1                                vfat        FAT32             2D1A-7E15                               299M     0% /boot/efi
[... 483 lines cut ...]
(15/15) Rebuilding certificate stores...
 2020-05-31 09:22:47 DEBUG alma                          > fstab:
UUID=55c64c4f-50a8-48c7-a099-0f8676e4c4a8	/         	ext4      	rw,noatime	0 1

UUID=6E79-AC43      	/boot     	vfat      	rw,noatime,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,utf8,errors=remount-ro	0 2

 2020-05-31 09:22:47 INFO  alma                          > Performing post installation tasks
[... cut ...]
 2020-05-31 09:22:50 DEBUG alma::storage::mount_stack    > Unmounting /tmp/.tmptABGD1/boot
 2020-05-31 09:22:56 DEBUG alma::storage::mount_stack    > Unmounting /tmp/.tmptABGD1
 2020-05-31 09:24:11 DEBUG alma::storage::crypt          > Closing encrypted device alma_root
 2020-05-31 09:24:11 ERROR alma                          > Initramfs error
 2020-05-31 09:24:11 ERROR alma                          > Caused by: exit code: 1

Afaik the is no `linux` package in manjaro, the kernel management is different

jamesmcm, 2020-05-31:
@Dorifer can you please check what files you do have in /etc/mkinitcpio.d ?

```
ls /etc/mkinitcpio.d/
```

It wants the `linux.preset` file which is installed by the `linux` package on Arch Linux. At least if Manjaro provides a different one you might be able to create a symbolic link to it.

As it seems Manjaro handles it differently: https://forum.manjaro.org/t/running-mkinitcpio-gives-me-this/25827

But if you symlink linux56.preset (or whatever your kernel version is) to linux.preset, it should work like Arch hopefully :slightly_smiling_face:

Dorifer, 2020-05-31:
I think you're onto something :) :

`ls /etc/mkinitcpio.d/`
`linux54.preset  linux54-rt.preset  linux55.preset`

jamesmcm, 2020-05-31:
Cool, so try:

```
sudo ln -s /etc/mkinitcpio.d/linux55.preset /etc/mkinitcpio.d/linux.preset
```

This isn't easy to support in ALMA since we can't assume which kernel version to check (maybe we could just list them and choose the highest version), but at least we could add a better error message.

Dorifer, 2020-06-02:
weird, I created the symbolic link to my current kernel preset:

> dorifer@dorifer-x220:/etc/mkinitcpio.d$ ls -l
> insgesamt 20
> -rw-r--r-- 1 root root 392 29. Dez 09:33 linux54.preset
> -rw-r--r-- 1 root root 404 22. Feb 14:25 linux54-rt.preset
> -rw-r--r-- 1 root root 392 24. Feb 15:05 linux55.preset.pacsave
> -rw-r--r-- 1 root root 392  2. Jun 10:21 linux56.preset
> -rw-r--r-- 1 root root 404  2. Jun 10:21 linux56-rt.preset
> lrwxrwxrwx 1 root root  32  2. Jun 14:02 linux.preset -> /etc/mkinitcpio.d/linux56.preset

but the script still doesnt recognize it:

> 
>  2020-06-02 14:39:14 INFO  alma > Generating initramfs
> /usr/bin/mkinitcpio: line 265: /etc/mkinitcpio.d/linux.preset: No such file or directory
> ==> ERROR: Failed to load preset: `/etc/mkinitcpio.d/linux.preset'
>  2020-06-02 14:40:44 ERROR alma > Initramfs error
>  2020-06-02 14:40:44 ERROR alma > Caused by: exit code: 1

jamesmcm, 2020-06-02:
Note that the command is run inside the chroot, so it needs to exist there.

The question is why pacstrap isn't copying it on install though.

Could you try with `alma-git` or building the repo with cargo + Rust?

jkerdreux-imt, 2020-06-03:
I run into the same issue. On manjaro, pacstrap doesn't provide any /etc/mkinitcpio.d/linux.preset, only linux56.preset for kernel 5.6 in my case. 

Is there a way have access to the chroot before image build process to fix that ?

jkerdreux-imt, 2020-06-03:
I patched alma to give me a shell just after pacstrap, created the sysmlink in the shell and exit. This fixed the issue.

jamesmcm, 2020-06-03:
So is there any way we can get Manjaro to create the normal mkinitcpio preset when installing via pacstrap?

Maybe sym-linking the kernel image itself prior to running ALMA? It'd be nice if we had like one command Manjaro users need to run prior to using ALMA to address this issue.

We could add an argument to go to an interactive chroot prior to building for sure, like a debug chroot.

r-darwish, 2020-06-03:
Sorry, but at this moment I don't intend to support Arch Linux derivatives, only Arch Linux itself.

closed 2020-06-03 by r-darwish