r-darwish/alma #51: any missing packages for USB media ?

omeringen, 2020-05-25:
Hi,
Sorry if it's irrelevant but i am newbie. I was slacking on Wiki and saw this :

> https://wiki.archlinux.org/index.php/Microcode#Installation
If your Arch installation is on a removable drive that needs to have microcode for both manufacturer processors, install both packages. 

Do we need this ? I couldn't find anything related to it on search in this repo.

jamesmcm, 2020-05-25:
`intel-ucode` is included: https://github.com/r-darwish/alma/blob/c8b151fe5ff5abe18a091e78567f3ed84c7470d5/src/constants.rs#L16

Though we should probably add AMD's one there too.

omeringen, 2020-05-25:
Okay thanks ;)

jamesmcm, 2020-05-25:
I created the PR #52 

Note if you are using your own preset files (you can see [my example here](https://github.com/jamesmcm/arch-i3-usb)), then you can just add this package to be installed with the others you install :slightly_smiling_face:

closed 2020-06-17 by r-darwish