r-darwish/alma #53: build from master fails

Dorifer, 2020-05-28:
Because of my problems from #49 I tried building the git version via the AUR:

==> Beginne build()...
   Compiling libc v0.2.67
   Compiling cfg-if v0.1.9
   Compiling lazy_static v1.4.0
   Compiling proc-macro2 v1.0.9
   Compiling unicode-xid v0.2.0
   Compiling syn v1.0.16
   Compiling memchr v2.2.1
   Compiling version_check v0.9.1
   Compiling proc-macro2 v0.4.30
   Compiling cc v1.0.40
   Compiling unicode-xid v0.1.0
   Compiling regex-syntax v0.6.15
   Compiling bitflags v1.1.0
   Compiling log v0.4.8
   Compiling ppv-lite86 v0.2.5
   Compiling syn v0.15.44
   Compiling unicode-width v0.1.5
   Compiling quick-error v1.2.2
   Compiling failure_derive v0.1.6
   Compiling termcolor v1.0.5
   Compiling rustc-demangle v0.1.16
   Compiling unicode-segmentation v1.3.0
   Compiling serde v1.0.98
   Compiling ansi_term v0.11.0
   Compiling remove_dir_all v0.5.2
   Compiling vec_map v0.8.1
   Compiling strsim v0.8.0
   Compiling nix v0.17.0
   Compiling void v1.0.2
   Compiling byte-unit v3.0.1
   Compiling thread_local v1.0.1
   Compiling proc-macro-error-attr v0.4.10
   Compiling proc-macro-error v0.4.10
   Compiling c2-chacha v0.2.2
   Compiling textwrap v0.11.0
   Compiling humantime v1.3.0
   Compiling heck v0.3.1
error: failed to run custom build command for `proc-macro-error-attr v0.4.10`

Caused by:
  process didn't exit successfully: `/var/tmp/pamac-build-dorifer/alma-git/src/alma/target/release/build/proc-macro-error-attr-248a21d7ac6b2a8c/build-script-build` (exit code: 101)
--- stderr
thread 'main' panicked at 'called `Option::unwrap()` on a `None` value', /home/dorifer/.cargo/registry/src/github.com-1ecc6299db9ec823/proc-macro-error-attr-0.4.10/build.rs:2:27
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace

warning: build failed, waiting for other jobs to finish...
error: build failed

jamesmcm, 2020-05-28:
This is due to a bad part of non-forwards compatibility in proc-macro-error-attr: https://gitlab.com/CreepySkeleton/proc-macro-error/-/blob/master/proc-macro-error-attr/build.rs

We'll need to bump up the structopt version in the Cargo.toml

Dorifer, 2020-05-29:
> Even in the above case it shouldn't fail. Do you have rust installed normally, are you able to run `rustc --version` ?

yes (rustc 1.43.1)

> Are you building on an unusual platform, like a musl based system or Arch Linux ARM?

Im on manjaro trying to build the aur package alma-git

jamesmcm, 2020-05-29:
The failing line is: 
https://gitlab.com/CreepySkeleton/proc-macro-error/-/blob/3e4223b3c5d2b815ad455b6df5099b3ea7d5a6a0/build.rs#L2

Bumping up the structopt version might still fix this (even though getting the version shouldn't be failing anyway), so I'll test it and create a PR later.

r-darwish, 2020-05-31:
I hope @jamesmcm fix will fix this. However, this tool isn't supported in Manjaro.

closed 2020-05-31 by r-darwish