r-darwish/alma #61: Consider using paru instead of yay

lnicola, 2020-11-23:
It's written by one of the `yay` authors and has binary releases, so it doesn't need to be compiled, at least on `x86-64` (but `yay` has those too).

jamesmcm, 2020-11-23:
Yeah, `paru-bin` could work well. It should be a simple change.

r-darwish, 2020-11-23:
To quote Morganamilo in Reddit:

> Jguer is still there, so there's no need to panic and move away from yay. Just don't expect much new development on it.

I'm not convinced that I should switch to paru while yay is still actively developed just because now it's the new cool kid in the block. When paru reaches the maturity level of yay, which so far never disappointed me, then I'll consider it.

closed 2020-11-23 by r-darwish

lnicola, 2020-11-24:
One issue with the way `yay` is built from source by `alma` is that `go` remains installed in the final image, currently at a hefty 560 MB.

jamesmcm, 2020-11-24:
Yeah, I'd recommend adding an extra field to the AurHelper struct for the specific package name to use, and then make yay-git the default, and remove go in the AUR_DEPENDENCIES :slightly_smiling_face: 

so:

https://github.com/r-darwish/alma/blob/c04b5f555989b8b41894f79d4a9b5a4d1618e11f/src/aur.rs#L5 - add package field here

https://github.com/r-darwish/alma/blob/c8b151fe5ff5abe18a091e78567f3ed84c7470d5/src/args.rs#L82 - allow specifying specific package name (though we need to get the provided command name somehow, this is in the PKGBUILD though)

https://github.com/r-darwish/alma/blob/8c2d7e6d37cbac81dda45f4b7d4735faace2a4f3/src/main.rs#L309 - don't use the direct name here, but the package name.

https://github.com/r-darwish/alma/blob/97f435bfbeed897034afa799a4d3f4abb6d4b3ff/src/constants.rs#L23 - remove go here

Using the binary packages definitely makes sense given the use case.

r-darwish, 2020-11-26:
I pushed a commit with your suggestion