r-darwish/alma #64: Can this tool be used to build what will become a read-only ISO for future Live Boots? (not persistent)

DwarfFighterCleric, 2020-11-28:
Man, from what I am reading, this tool seems to be awesome! Thanks!  
I am searching for a way to create a "respin" or a "snapshot" of my system, compress it down into SquashFS and run it as Live-Session to use the least disk writes as possible to my SSD drive. (I guess the better phrasing of it would be "a custom Live-Session ISO" with all my custom apps on it)  
So the "persistent" side of it is not of interest to me. Also, I like very much having a working functional system at every reboot, without having to worry about breakages, package manager becoming stuck, and so on. Reboot, it's back to stage zero.  
Does this tool uses the Chroot method for customizations? Do you know of any chroot tools or methods to achieve what I'm looking for out there?  
I used to use a tool called JLiveCD for Ubuntu-based distros, but because of my new Ryzen hardware I am pretty much "locked" down with Manjaro, because it was the only one that worked properly from 1st boot, and that tool doesn't work for Manjaro yet, there are a few specific quirks on it that prevent it.

DwarfFighterCleric, 2020-11-28:
Or even better: is there a way to use this tool and "bypass" the Persistence parts of it somehow?  
I would need to have it locked down Read-Only.  
Extremely unlikely, but I'm asking it anyway.

r-darwish, 2020-11-28:
Manjaro isn't supported in ALMA. It might work by accident but I will not fix it if it breaks, so I don't think this is the proper tool to rely on if you wish to use Manjaro instead of Arch Linux.

Having said that, you can enable [systemd volatile boot](https://www.freedesktop.org/software/systemd/man/systemd-volatile-root.service.html)

closed 2020-12-03 by r-darwish