r-darwish/alma #67: I want to use grub to boot the img image created by ALMA.

daiaji, 2020-12-05:
Use similar grub commands.
```
menuentry "Manjaro img" {
    insmod exfat
    set isofile="/Linux/manjaro.img"
    set dri="free"
    search --no-floppy -f --set=root $isofile
    probe -u $root --set=uuid
    set dev="/dev/disk/by-uuid/$uuid"
    loopback loop $isofile
    linux  (loop)/boot/vmlinuz-5.9-x86_64 img_dev=$dev img_loop=$isofile driver=$dri tz=utc keytable=us
    initrd  (loop)/boot/amd_ucode.img (loop)/boot/intel_ucode.img (loop)/boot/initramfs-5.9-x86_64.img
}
```
However, the current img generated by ALMA seems to lack adaptation to this.

r-darwish, 2020-12-05:
First of all, Manjaro isn't supported in ALMA. If it works then have fun, if it doesn't then I won't fix it.

Second, I don't understand what you are asking. ALMA is intented to directly work on a USB device, so it will fill up the entire space. It supports image creation, but only raw fixed-sized images.

daiaji, 2020-12-06:
@r-darwish I just want to use gurb2 to boot the img image generated by ALMA instead of letting ALMA use the entire USB device.

r-darwish, 2020-12-06:
Can't you just create an image with the `--image` flag?

daiaji, 2020-12-08:
I can create an img file with the --image tag.
But only the img file that can be used in qemu is useless to me.

r-darwish, 2020-12-08:
I'm going to close this one as it currently isn't a scenario I wish to support

closed 2020-12-08 by r-darwish