r-darwish/alma #71: Did alma just kill my main system?

TheGamingCoderNoob, 2021-04-27:
**Describe the bug** 
A clear and concise description of what the bug is.
I made an ALMA usb, booted into it and now my main system isn't bootable. I didn't even login!

**To Reproduce**
Steps to reproduce the behavior:
Idk it just happened

**Expected behavior**
A clear and concise description of what you expected to happen.
It should in no way be able to do this

**Additional context**
Add any other context about the problem here.

TheGamingCoderNoob, 2021-04-27:
I have.

r-darwish, 2021-04-27:
There's no bug I can find with "Idk it just happened" as the only information

TheGamingCoderNoob, 2021-04-28:
Update: It turns out my boot partition was wiped so i had to reinstall kernel and grub then regen fstab

TheGamingCoderNoob, 2021-04-28:
It is still worth looking into cause this may have been a bug.

r-darwish, 2021-04-29:
Unless you can provider more information I probably won't be able to guess it. 
* What was the name of the disk with your main system? Is this a removable device?
* Is the boot partition stored on that same disk? 
* Which bootloader are you using?
* EFI or BIOS?
* What was the name of the disk that you ran ALMA on?
* Did you use `--allow-non-removable?
* What exactly happened to your boot partition? Did it contain an empty file system or was there no file system

TheGamingCoderNoob, 2021-04-29:
1. /dev/sda, not removable.
2. Yes,at /dev/sda1.
3. GRUB
4. UEFI
5. /dev/sdb
6. No
7. Empty FS

matidfk, 2022-06-07:
This has just happened to me aswell - I'll answer the same questions as KSPAtlas
1. /dev/sda, not removable
2. Yes, I believe /dev/sda1 or 3
3. grub
4. uefi
5. /dev/sdc
6. no
7. empty filesystem