r-darwish/alma #80: [Error]/[Question] alma failing on MBR partition device.

HarshalRathore, 2021-12-03:
Hey there!
My laptop is MBR partitioned and uses Legacy BIOS not UEFI. 
I ran this command `ALMA_USER="harshal" TIMEZONE="Asia/Kolkata" sudo -E alma create --presets ./arch-i3-usb/preset`
I knowingly did not passed the USB address because it is said in documentation that alma prompts the user for removable device address if none is passed.
below is the  error that I got with above command.
```
***************************************************************
Found invalid GPT and valid MBR; converting MBR to GPT format
in memory.
***************************************************************

Warning: The kernel is still using the old partition table.
The new table will be used at the next reboot or after you
run partprobe(8) or kpartx(8)
GPT data structures destroyed! You may now partition the disk using fdisk or
other utilities.
Warning: The kernel is still using the old partition table.
The new table will be used at the next reboot or after you
run partprobe(8) or kpartx(8)
The operation has completed successfully.
 2021-12-03T07:46:07.571Z INFO  alma > Formatting filesystems
mkfs.fat 4.2 (2021-01-31)
mkfs.fat: /dev/sdb1 contains a mounted filesystem.
Error: Error formatting filesystem

Caused by:
    Bad exit code: exit status: 1
```

Question: Does alma do not work on MBR styled devices with Legacy BIOS?
Thank you 4 this awesome tool.

r-darwish, 2021-12-03:
Alma makes disks which are bootable both in legacy and UEFI mode. Is it possible that the disk you tried to reformat was already mounted?

HarshalRathore, 2021-12-03:
@r-darwish First thanks for the fast response.
> Is it possible that the disk you tried to reformat was already mounted?

Yes I checked in my File manager it was mounted and I could visit that file system.

HarshalRathore, 2021-12-03:
The problem has been solved. It required a reboot after doing that it is working.

closed 2021-12-03 by HarshalRathore