r-darwish/alma #85: [Question] Which package is automatically installed and did not need to be added to preset files?

LazyGeniusMan, 2022-01-20:
Hi, I was wondering where I can find the list of package that will automatically installed so it doesn't need to be added to preset files? at first I thought it's package from `archiso` but after I use ALMA looks like it is not? there are some package missing compared to `archiso` package list from [Arch Gitlab](https://gitlab.archlinux.org/archlinux/archiso/-/blob/master/configs/releng/packages.x86_64). I use `user` preset and `kde` preset btw

jamesmcm, 2022-01-20:
See https://github.com/r-darwish/alma/blob/master/src/constants.rs

```
    "base",
    "linux",
    "linux-firmware",
    "grub",
    "efibootmgr",
    "intel-ucode",
    "networkmanager",
    "broadcom-wl",
    "amd-ucode",
```

LazyGeniusMan, 2022-01-20:
aight thanks

closed 2022-01-20 by LazyGeniusMan