r-darwish/alma #86: external USB drive error

ng0177, 2022-04-10:
Although possible make this external USB drive bootable with Arch, the below error is issued:
```
$ ls -al /dev/disk/by-id/
lrwxrwxrwx 1 root root   9 10. Apr 16:24 usb-Seagate_BUP_Slim_BK_NA97E1B5-0:0 -> ../../sda

sudo ALMA_USER=archie alma create /dev/disk/by-id/usb-Seagate_BUP_Slim_BK_NA97E1B5-0:0 --presets ./user.toml

2022-04-10 16:26:46 ERROR alma > The given block device is neither removable nor a loop device
```

jamesmcm, 2022-04-10:
What does it have under the `/sys/block/{name}/` entry? Is `removable` there?

Otherwise you can use `--allow-non-removable` - make sure you specify the correct device though!

ng0177, 2022-04-10:
`--allow-non-removable` works like a charm.
```
$ ls -al /sys/block/
lrwxrwxrwx  1 root root 0 10. Apr 20:49 sda -> ../devices/pci0000:00/0000:00:14.0/usb4/4-1/4-1:1.0/host0/target0:0:0/0:0:0:0/block/sda
```

ng0177, 2022-04-10:
I still need to figure out how to increase the font size on a 4K monitor, though, to be able use it efficiently.

jamesmcm, 2022-04-10:
Check https://wiki.archlinux.org/title/HiDPI it's per-application for most things unfortunately.

ng0177, 2022-04-11:
Only terminal i.e. `console tty` are relevant to me as I have not tried to get e.g. `Gnome` to work with `alma` (would be nice).

closed 2022-04-12 by ng0177